v923z/micropython-ulab #554: Issue opening a CPython *.npy file

A622266, 2022-10-10:
**Description**
In ulab, I cannot open a *.npy file that was created in CPython.

**Background**
I want to use the ulab function 'sosfilt' on a data stream.  I use CPython filter design tools found in Scipy to create an ndarray describing second order sections.  I use CPython Numpy function 'save' to save the file  in *.npy format.  However, when I try to open this file in ulab using ulab function 'load', I get an error indicating that ulab cannot open the file.  I have verified that the data saved using CPython 'save' is in ndarray format by using the 'type' function.  Of course I can make progress by simply print the output of the CPython filter design tool and creating an ndarray in ulab.  But I am hoping to have a more automated process for rapid experimenting. Is it currently possible?

**To reproduce**
Run following code with attached sos.npy (after unzipping)
```python
import ulab
from ulab import numpy as np
from ulab import scipy as spy
print('version string: ', ulab.__version__)

np.load('sos.npy')
```
**Output**
version string:  6.0.0-2D-c
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
OSError: [Errno 2] ENOENT

**Expected behavior**
Open 'sos.npy' and recognize it as ndarray in ulab
[sos.zip](https://github.com/v923z/micropython-ulab/files/9749238/sos.zip)

v923z, 2022-10-11:
Before I can fix the issue, there are two ways for you to proceed: first is to simply save your data in a text file with `savetxt`, and then load the contents text file with `loadtxt`. In fact, you can even do something very similar in `python`, you don't even need the `savetxt/loadtxt` functions. Writing your own data reader in `python` is probably safer, because you can manipulate the exact data format in a much more flexible way.

However, if you know what type of numbers you except, you could even save save them in a binary format, and then fill up your `ndarray` from a binary string.

But I have the feeling that the issue is that your numbers are of type `float64`, and `ulab` most probably doesn't support that on the ESP32. You could find this out by looking at the output of `.itemsize` in CPython, and `micropython`, or simply 

```
from ulab import numpy as np

a = np.array([1, 2, 3])
print(a)
```

A622266, 2022-10-12:
Yes you are right the data type is ```float64``` from ```numpy```, and so that probably explains why it doesn't work on esp32.  I'll try to come up with a ```python``` solution, and avoid use of ```savetxt/loadtxt``` as you suggest.

v923z, 2022-10-12:
You can explicitely cast your results to `float32` in `numpy` before calling the `save` function. 

But this still needs to be fixed, because the `load` function should simply bail out, of the format is not supported, instead of crashing.