v923z/micropython-ulab #555: [BUG] TypeError with arange function

A622266, 2022-10-17:
**Describe the bug**
Call to `arange` results in `TypeError` in `ulab`.  Same call gives correct result in `Cpython`

```python
import ulab
from ulab import numpy as np
print('version string: ', ulab.__version__)
print('You are running ulab')

t = np.arange(start=0,stop=0.01,step=0.01/8)
print(t)
```

**To Reproduce**
Run above code in ulab (esp32 s3) and get the following result:
version string:  6.0.0-2D-c
You are running ulab
Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
TypeError: function missing 1 required positional arguments

**Expected behavior**
from `CPython`:
[0.      0.00125 0.0025  0.00375 0.005   0.00625 0.0075  0.00875]

v923z, 2022-10-18:
Thanks for bringing this up! I'll look into it.

v923z, 2022-10-28:
@A622266 Could you, please, list all possible combinations of the arguments of `arange`? That function is a very strange one, and sort of buggy, even in `numpy`.

By the way, 

```
t = np.arange(0, 0.01, 0.01/8)
```
should work.

A622266, 2022-10-29:
You are right, it does work.  The issue is apparently the keywords (eg. ```start=```).   ```CPython``` allows them, even though I see now that they are not called for in the [function prototype](https://numpy.org/doc/stable/reference/generated/numpy.arange.html).  I like them because it is a little more clear, but I can always add a comment.  So perhaps we should consider this topic closed?

v923z, 2022-10-29:
I think you encounter the problem only, if you supply all three arguments via keywords. What makes the implementation a bit difficult is the fact that the order of argument changes in an inconsistent way: 
```
arange(stop)
arange(start, stop)
arange(start, stop, step)
```
Since the function works without keywords, and flash space is limited on microcontrollers, I would rather close the issue.

A622266, 2022-10-30:
Yes I agree

closed 2022-10-30 by A622266