v923z/micropython-ulab #556: Missing Step for RP2 Board Installation

Darkai87, 2022-10-27:
Hello I don't know if this issue is on my end or not but I've been struggling to install ulab onto my raspberry pi pico using the instructions found under RP2-Based Boards. When I follow the download process everything seems to go well until I need to compile mpy-cross through the use of the "make" command. Here is the following error I receive in response to executing the command in the command prompt shell:

----------------------------------------------------------------------------------------------------

C:\Users\cycri\pico-project\micropython\mpy-cross>make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
process_begin: CreateProcess(NULL, uname -s, ...) failed.
Makefile:10: pipe: No error
process_begin: CreateProcess(NULL, gcc -dumpmachine, ...) failed.
Makefile:54: pipe: No error
process_begin: CreateProcess(NULL, gcc -dumpmachine, ...) failed.
../py/mkrules.mk:197: pipe: No error
"GEN build/genhdr/qstr.i.last"
Traceback (most recent call last):
  File "C:\Users\cycri\pico-project\micropython\py\makeqstrdefs.py", line 210, in <module>
    preprocess()
  File "C:\Users\cycri\pico-project\micropython\py\makeqstrdefs.py", line 76, in preprocess
    for output in p.imap(pp(flags), chunks):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\multiprocessing\pool.py", line 873, in next
    raise value
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\multiprocessing\pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "C:\Users\cycri\pico-project\micropython\py\makeqstrdefs.py", line 60, in run
    return subprocess.check_output(args.pp + flags + files)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 503, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 971, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2288.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 1440, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
make: *** [../py/mkrules.mk:111: build/genhdr/qstr.i.last] Error 1
make: *** Deleting file 'build/genhdr/qstr.i.last'

--------------------------------------------------------------------------------------------------
 
Hope this error code helps in troubleshooting my issue, but just in case this is an error due to a missing step under the README I figured I'd leave this bug report. Any help is greatly appreciated.

Thanks.

v923z, 2022-10-27:
It seems to me that this is an issue with 

https://github.com/v923z/micropython-builder

Darkai87, 2022-10-27:
Okay I understand thank you for the quick response and yes I'm running the Windows OS, also where should the python directory be located when I execute the make command?

Darkai87, 2022-10-27:
So I've also tried the micropython-builder method of downloading ulab however still when it gets to the mpy-cross file to make the makefile I got the same error and the configuration fails. Any suggestions?

v923z, 2022-10-28:
What I meant is that you can download the firmware directly from https://github.com/v923z/micropython-builder/releases. If your particular board is not listed there, then we can easily add that, and the firmware will be compiled for you every second day.

ktulukid, 2022-11-09:
Just go to https://github.com/v923z/micropython-builder/releases, download 'PICO.uf2', boot your pico while holding reset, drop 'PICO.uf2' onto the RPI-RP2 volume instead of the 'MicroPython UF2' that the start up guide tells you to use.

v923z, 2022-11-29:
@Darkai87 I don't think that the issue is still relevant, so closing now.

closed 2022-11-29 by v923z