v923z/micropython-ulab #566: Dimension remains constant despite editing `ULAB_MAX_DIMS` in `ulab.h`.

HaydenS3, 2023-01-03:
**Describe the bug**

Dimension remains 2  despite editing `ULAB_MAX_DIMS` to 3 in `ulab.h`. Compliling with the `build.sh` script.

ulab version: 6.0.1

**To Reproduce**

1. Modify `ULAB_MAX_DIMS` in `ulab.h` to eqaul 3
2. `./build.sh`
3. Run the following with `ulab/micropython/ports/unix/build-2/micropython-2`

```
import ulab
print(ulab.__version__)
```

Output: `6.0.1-2D-c`

**Expected behavior**

Output: `6.0.1-3D-c`

**Additional context**

Not sure if this is a bug or if I'm compiling incorrectly. Any help is appreciated!

v923z, 2023-01-03:
I don't think this is a bug: the string is generated in the lines https://github.com/v923z/micropython-ulab/blob/1a440d7d124a20a3dbaa0909000741eb28a623b3/code/ulab.c#L40-L44

so, there is really nothing that could go wrong there. But I believe I know why you have the difficulty: builds are placed in different directories according to their dimension. Thus, 

```
Run the following with ulab/micropython/ports/unix/build-2/micropython-2
```
means that you are still running the 2-dimensional binary. You should have 
```
ulab/micropython/ports/unix/build-3/micropython-3
```
which is the correct place for three dimensions.

HaydenS3, 2023-01-03:
@v923z That makes sense, however after running `build.sh` with the modified `ULAB_MAX_DIMS` no `unix/build-3` directory is created.

Here's an output from the `build.sh` script for reference: [https://pastebin.com/CUPG0Pq9](https://pastebin.com/CUPG0Pq9)

v923z, 2023-01-03:
You pass the dimension here: https://github.com/v923z/micropython-ulab/blob/1a440d7d124a20a3dbaa0909000741eb28a623b3/build.sh#L44

So, you would call `./build.sh` as `./build.sh 3`, if you want to have three dimensions. You can also see that here: https://github.com/v923z/micropython-ulab/blob/1a440d7d124a20a3dbaa0909000741eb28a623b3/.github/workflows/build.yml#L56

HaydenS3, 2023-01-03:
@v923z Created a PR with a quick change to the readme for clarification. Is there anywhere else this should be mentioned in the docs or anything else that should be changed?

v923z, 2023-01-03:
@HaydenS3 Many thanks for https://github.com/v923z/micropython-ulab/pull/567. It's been merged.

closed 2023-01-03 by v923z