v923z/micropython-ulab #570: make error PYBLITEV10

Thomaspiketty, 2023-01-05:
I tried to make Firmware for **pyboard lite v1.0**

I just followed the README.md **STM-based boards**

**I typed:**
pi@raspberrypi:~/micropython/ports/stm32$ **make BOARD=PYBLITEV10 USER_C_MODULES=../../../ulab all**

**and It gives me an error:**
-e Error: micropython-lib submodule is not initialized. Run 'make submodules'
make: *** [../../py/mkrules.mk:188: build-PYBLITEV10/frozen_content.c] Error 1

**So I typed:** pi@raspberrypi:~/micropython/ports/stm32$ **make submodules**

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/libhydrogen lib/stm32lib lib/micropython-lib
Synchronizing submodule url for '../../lib/stm32lib'
Submodule 'lib/libhydrogen' (https://github.com/jedisct1/libhydrogen.git) registered for path '../../lib/libhydrogen'
Submodule 'lib/micropython-lib' (https://github.com/micropython/micropython-lib.git) registered for path '../../lib/micropython-lib'
Cloning into '/home/pi/micropython/lib/libhydrogen'...
Cloning into '/home/pi/micropython/lib/micropython-lib'...
Submodule path '../../lib/libhydrogen': checked out '5c5d513093075f7245ea522101b17c50aa579af2'
Submodule path '../../lib/micropython-lib': checked out '038b4ac6572d0d5b4c18148dc7d7fdc026369fa4'

**I still couldn't see the firmware.dfu in Build-PYBLITEV10 directory. So I typed again :** 
pi@raspberrypi:~/micropython/ports/stm32$ **make BOARD=PYBLITEV10 USER_C_MODULES=../../../ulab all**

**And It gives me another error :** 
CC build-PYBLITEV10/pins_PYBLITEV10.c
LINK build-PYBLITEV10/firmware.elf
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: build-PYBLITEV10/firmware.elf section `.text' will not fit in region `FLASH_TEXT'
/usr/lib/gcc/arm-none-eabi/8.3.1/../../../arm-none-eabi/bin/ld: region `FLASH_TEXT' overflowed by 512 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:658: build-PYBLITEV10/firmware.elf] Error 1

**HOWEVER** When I tried PYBV10 I could generate the firmware.dfu without problem.

Tried OS : Raspberry pi 4 Bullseye, Macbook pro M1. And the errors are same.
Could you please help me?

v923z, 2023-01-22:
@Thomaspiketty Do you still want to keep the issue open?

Thomaspiketty, 2023-01-25:
sorry for the delay.
Deleting functions or reducing dimensions could not be an option for my project.
So I just decided to use another microcontroller.
I already had STM32 F401RE, and It works.
What I can't understand is, STM32 F401RE and Pyboard Lite have the same memory, If I correctly know.
But STM32 works and pyboard not...

I don't know that deleting some functions or reducing dimensions can be a solution. But if you want, you can close it.

v923z, 2023-01-25:
> sorry for the delay. Deleting functions or reducing dimensions could not be an option for my project. So I just decided to use another microcontroller. I already had STM32 F401RE, and It works. What [...]

The compiled firmware might not be of the same size.

closed 2023-01-25 by v923z