v923z/micropython-ulab #572: [BUG] crash when numpy.concatenate() given scalars

dhalbert, 2023-01-12:
ulab version is commit e68bb707 (current CircuitPython `main` version). `ulab.__version__` reports `6.0.2-2D`, but I think that commit is past that?

**To Reproduce**
```py
>>> import ulab
>>> ulab.numpy.concatenate((2.0,4.0,6.0), axis=1)
[crash]
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
```

**Expected behavior**
I believe the above is incorrect because `concatenate()` is being passed a sequence of scalars instead of a sequence of arrays. (This was [some user code](https://github.com/adafruit/circuitpython/issues/7446)). But it should not crash. CPython numpy reports an error:
```py
>>> numpy.concatenate((2.0,4.0,6.0), axis=1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<__array_function__ internals>", line 5, in concatenate
ValueError: zero-dimensional arrays cannot be concatenated
```
This is not a showstopper bug, since it involves incorrect usage.

v923z, 2023-01-12:
> ulab version is commit [e68bb70](https://github.com/v923z/micropython-ulab/commit/e68bb707b20ee326d84ab75fc9fb35f2e85b87e3) (current CircuitPython `main` version). `ulab.__version__` reports `6.0.2-2D`, but I think that commit is past that?

I don't know what you mean. The version number was bumped in that commit: https://github.com/v923z/micropython-ulab/commit/e68bb707b20ee326d84ab75fc9fb35f2e85b87e3

> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "<__array_function__ internals>", line 5, in concatenate
> ValueError: zero-dimensional arrays cannot be concatenated [...]

It should still be fixed, thanks for bringing it up!

dhalbert, 2023-01-12:
I meant that if click on "New Issue", and choose "Bug", the bug template says to fetch the version from `np.__version__`, instead of saying to use `ulab.__version__`. But `np.__version__` doesn't work, the latter does. So I think the issue template needs to be corrected: https://github.com/v923z/micropython-ulab/blob/master/.github/ISSUE_TEMPLATE/bug_report.md

v923z, 2023-01-14:
> I meant that if click on "New Issue", and choose "Bug", the bug template says to fetch the version from `np.__version__`, instead of saying to use `ulab.__version__`. But `np.__version__` doesn't [...]

OK, now I get it. Thanks for bringing this up! I've corrected the template.

v923z, 2023-01-14:
https://github.com/v923z/micropython-ulab/pull/575 fixes the issue. `numpy` allows array-likes in general ([concatenate](https://numpy.org/doc/stable/reference/generated/numpy.concatenate.html)), but here I would really like to keep this to `ndarray`s only, given that array-likes would give an overhead not only in terms of code, but RAM, too. This overhead would not be obvious to the user, which I would find disturbing on a small MCU. If one wants to concatenate lists, or tuples, they can still be explicitly cast like

```python
np.concatenate((np.array([1, 2]), np.array([3, 4]))
```
I hope this is a satisfactory solution.

closed 2023-01-15 by v923z