v923z/micropython-ulab #574: [BUG] Memory corruption with incorrect use of ulab.numpy.dot

jepler, 2023-01-14:
**Describe the bug**
```
Adafruit CircuitPython 8.0.0-beta.6 on 2022-12-21; Adafruit ItsyBitsy M4 Express with samd51g19
>>> ulab.__version__
'6.0.2-2D-c'
```
as well as "unix circuitpython" with the same ulab version

**To Reproduce**
```python
import ulab.numpy as np
coeff = np.ones((6,6))
values = np.ones(6) 
# prints array([6.0]) while python3 prints [6. 6. 6. 6. 6. 6.] 
print(np.dot(values, coeff))
# eventually causes crash!
while True: x = np.dot(values, coeff)
```

**Expected behavior**
Would like behavior to match standard Python & not hardfault

**Additional context**
Is an output array of wrong size being allocated?

v923z, 2023-01-14:
On the one hand, `dot` should be able to catch the wrong input type, so that issue should definitely be fixed. On the other hand, I have this: 

```python
>>> import ulab
>>> ulab.__version__
'6.0.2-2D-c'
>>> from ulab import numpy as np
>>> np.ones(6)
array([1.0, 1.0, 1.0, 1.0, 1.0, 1.0], dtype=float64)
```
So, I don't quite understand, why your very same version prints an array of length one. Even more perplexing is the fact that it prints the value 6.0. But I feel that even the `python3` version is wrong here: `ones` should produce, well, ones, and not sixes.

jepler, 2023-01-14:
[I changed the example so it tries to state more clearly that the shape of the result of `dot` is different]

PR #577 (2023-01-15): Allocate 1d results of correct size
Before, it was erroneously allocated as a 1-element array instead of a 6-element array in the test case.

Closes #574
commit 6e6d24ea (2023-01-15): Allocate 1d results of correct size (#577)

Before, it was erroneously allocated as a 1-element array
instead of a 6-element array in the test case.

Closes #574
closed 2023-01-15 by v923z