v923z/micropython-ulab #595: [BUG] Can't compile ulab with BOARD=GENERIC_S3 (GENERIC works fine)

zachmoshe, 2023-04-14:
I'm trying to compile `ulab` into MicroPython, following [this page](https://github.com/v923z/micropython-ulab#esp32-based-boards).  I have an old  ESP32 and a newer ESP32-S3 which I want to try. 

I've created two makefiles for `BOARD=GENERIC` and `BOARD=GENERIC_S3`. The first one works as expected, but when I try the S3 one I get this: 

```
zmoshe@laptop ~/proj/micropython/ports/esp32 % make -f makefile_ulab_s3
idf.py -D MICROPY_BOARD=GENERIC_S3 -D MICROPY_BOARD_DIR=/Users/zmoshe/proj/micropython/ports/esp32/boards/GENERIC_S3 -B build-GENERIC_S3  -DUSER_C_MODULES=/Users/zmoshe/proj/ulab/code/micropython.cmake build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Executing action: all (aliases: build)
Running ninja in directory /Users/zmoshe/proj/micropython/ports/esp32/build-GENERIC_S3
Executing "ninja all"...
[1/297] Performing build step for 'bootloader'
[1/1] cd /Users/zmoshe/proj/micropython/ports/esp32/build-GENERIC_S3/bootloader/esp-idf/esptool_py && /Users/zmoshe/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/zmoshe/proj/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 /Users/zmoshe/proj/micropython/ports/esp32/build-GENERIC_S3/bootloader/bootloader.bin
Bootloader binary size 0x48d0 bytes. 0x3730 bytes (43%) free.
[156/214] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/__/__/frozen_content.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/__/__/frozen_content.c.obj
/Users/zmoshe/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DFFCONF_H=\"/Users/zmoshe/proj/micropython/lib/oofatfs/ffconf.h\" -DLFS1_NO_ASSERT -DLFS1_NO_DEBUG -DLFS1_NO_ERROR -DLFS1_NO_MALLOC -DLFS1_NO_WARN -DLFS2_NO_ASSERT -DLFS2_NO_DEBUG -DLFS2_NO_ERROR -DLFS2_NO_MALLOC -DLFS2_NO_WARN -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DMICROPY_ESP_IDF_4=1 -DMICROPY_MODULE_FROZEN_MPY="(1)" -DMICROPY_PY_BTREE=1 -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_VFS_FAT=1 -DMICROPY_VFS_LFS2=1 -DMODULE_ULAB_ENABLED=1 -D__DBINTERFACE_PRIVATE=1 -Dvirt_fd_t="void*" -I/Users/zmoshe/proj/micropython/ports/esp32/build-GENERIC_S3/config -I/Users/zmoshe/proj/micropython -I/Users/zmoshe/proj/micropython/lib/berkeley-db-1.xx/PORT/include -I/Users/zmoshe/proj/ulab/code -I/Users/zmoshe/proj/micropython/ports/esp32 -I/Users/zmoshe/proj/micropython/ports/esp32/boards/GENERIC_S3 -I/Users/zmoshe/proj/micropython/ports/esp32/build-GENERIC_S3 -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble -I/Users/zmoshe/proj/esp-idf/components/newlib/platform_include -I/Users/zmoshe/proj/esp-idf/components/freertos/include -I/Users/zmoshe/proj/esp-idf/components/freertos/include/esp_additions/freertos -I/Users/zmoshe/proj/esp-idf/components/freertos/port/xtensa/include -I/Users/zmoshe/proj/esp-idf/components/freertos/include/esp_additions -I/Users/zmoshe/proj/esp-idf/components/esp_hw_support/include -I/Users/zmoshe/proj/esp-idf/components/esp_hw_support/include/soc -I/Users/zmoshe/proj/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/zmoshe/proj/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/zmoshe/proj/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/Users/zmoshe/proj/esp-idf/components/heap/include -I/Users/zmoshe/proj/esp-idf/components/log/include -I/Users/zmoshe/proj/esp-idf/components/lwip/include/apps -I/Users/zmoshe/proj/esp-idf/components/lwip/include/apps/sntp -I/Users/zmoshe/proj/esp-idf/components/lwip/lwip/src/include -I/Users/zmoshe/proj/esp-idf/components/lwip/port/esp32/include -I/Users/zmoshe/proj/esp-idf/components/lwip/port/esp32/include/arch -I/Users/zmoshe/proj/esp-idf/components/soc/include -I/Users/zmoshe/proj/esp-idf/components/soc/esp32s3/. -I/Users/zmoshe/proj/esp-idf/components/soc/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/hal/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/hal/include -I/Users/zmoshe/proj/esp-idf/components/hal/platform_port/include -I/Users/zmoshe/proj/esp-idf/components/esp_rom/include -I/Users/zmoshe/proj/esp-idf/components/esp_rom/include/esp32s3 -I/Users/zmoshe/proj/esp-idf/components/esp_rom/esp32s3 -I/Users/zmoshe/proj/esp-idf/components/esp_common/include -I/Users/zmoshe/proj/esp-idf/components/esp_system/include -I/Users/zmoshe/proj/esp-idf/components/esp_system/port/soc -I/Users/zmoshe/proj/esp-idf/components/esp_system/port/public_compat -I/Users/zmoshe/proj/esp-idf/components/xtensa/include -I/Users/zmoshe/proj/esp-idf/components/xtensa/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/driver/include -I/Users/zmoshe/proj/esp-idf/components/driver/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/esp_pm/include -I/Users/zmoshe/proj/esp-idf/components/esp_ringbuf/include -I/Users/zmoshe/proj/esp-idf/components/efuse/include -I/Users/zmoshe/proj/esp-idf/components/efuse/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/vfs/include -I/Users/zmoshe/proj/esp-idf/components/esp_wifi/include -I/Users/zmoshe/proj/esp-idf/components/esp_event/include -I/Users/zmoshe/proj/esp-idf/components/esp_netif/include -I/Users/zmoshe/proj/esp-idf/components/esp_eth/include -I/Users/zmoshe/proj/esp-idf/components/tcpip_adapter/include -I/Users/zmoshe/proj/esp-idf/components/esp_phy/include -I/Users/zmoshe/proj/esp-idf/components/esp_phy/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/esp_ipc/include -I/Users/zmoshe/proj/esp-idf/components/app_trace/include -I/Users/zmoshe/proj/esp-idf/components/esp_timer/include -I/Users/zmoshe/proj/esp-idf/components/app_update/include -I/Users/zmoshe/proj/esp-idf/components/spi_flash/include -I/Users/zmoshe/proj/esp-idf/components/bootloader_support/include -I/Users/zmoshe/proj/esp-idf/components/bt/common/osi/include -I/Users/zmoshe/proj/esp-idf/components/bt/include/esp32s3/include -I/Users/zmoshe/proj/esp-idf/components/bt/common/api/include/api -I/Users/zmoshe/proj/esp-idf/components/bt/common/btc/profile/esp/blufi/include -I/Users/zmoshe/proj/esp-idf/components/bt/common/btc/profile/esp/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/porting/nimble/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/port/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/dis/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gap/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/gatt/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ias/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ipss/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/lls/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/tps/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/util/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/ram/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/include -I/Users/zmoshe/proj/esp-idf/components/bt/host/nimble/esp-hci/include -I/Users/zmoshe/proj/esp-idf/components/nvs_flash/include -I/Users/zmoshe/proj/esp-idf/components/mbedtls/port/include -I/Users/zmoshe/proj/esp-idf/components/mbedtls/mbedtls/include -I/Users/zmoshe/proj/esp-idf/components/mbedtls/esp_crt_bundle/include -I/Users/zmoshe/proj/esp-idf/components/esp_adc_cal/include -I/Users/zmoshe/proj/esp-idf/components/mdns/include -I/Users/zmoshe/proj/esp-idf/components/console -I/Users/zmoshe/proj/esp-idf/components/sdmmc/include -I/Users/zmoshe/proj/esp-idf/components/ulp/include -I/Users/zmoshe/proj/esp-idf/components/tinyusb/tinyusb/src -I/Users/zmoshe/proj/esp-idf/components/tinyusb/additions/include -I/Users/zmoshe/proj/esp-idf/components/freertos/include/freertos -mlongcalls -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error= [... cut ...]
 #define QDEF(id, hash, len, str) id,
                                  ^~
[173/214] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/Users/zmoshe/proj/ulab/code/numpy/numerical.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [all] Error 1
```

I'm a bit clueless with C compilation errors. Looks like something is declared twice but not sure why this doesn't happen for the regular board type.. Any ideas?

v923z, 2023-04-14:
What happens if you remove the artifacts, and start from scratch?

hamza-712, 2023-04-14:
I was having the same problem. I used esp idf v4.4.3 and I was able to build firmware.

hamza-712, 2023-04-14:
@v923z I need help too. I have built firmware for ESP-WROOM-32 but my ESP32 keeps **resetting** indefinitely. 
Is there a compiled binary available with latest micropython firmware?

I looked at the troubleshooting link as well. I changed CONFIG_ESPTOOLPY_FLASHMODE but it didn't work

zachmoshe, 2023-04-14:
OK, it's a bit weird but I can confirm it works for me..
It didn't at first, then as Zoltan suggested I tried `make clean` and even re-cloning micropython from scratch. After a few tries I was able to compile both the GENERIC and the GENERIC_S3 boards. I'm not sure what exactly did the trick but here are the settings that work for me: 

* I have micropython at commit `0acc73344` (this is currently origin/master). I only added 2 files in the `ports/esp32` folder called `makefile_ulab` and `makefile_ulab_s3` with the following content: 
```
BOARD = GENERIC   # GENERIC_S3 in the s3 file.
USER_C_MODULES = <PATH TO ULAB FOLDER>/code/micropython.cmake

include Makefile
```
* I have ulab at commit `8585407` (one before the latest, don't think it will make a difference). The only change I have there is in `code/ulab.h`: 
```
-#define ULAB_MAX_DIMS                       2
+#define ULAB_MAX_DIMS                       4
```
* My ESP-IDF is `v4.4`.

From the `micropython/ports/esp32` I can successfully run `make -f makefile_ulab_s3` now (for both files). 
I also uploaded the result binaries to ESP32-S3 and it seems to work (was able to import numpy form the repl, didn't check further).

@hamza-712 - Hope this helps...

hamza-712, 2023-04-15:
thanks i have successfully built firmware
I use idf.py command to build firmware instead of make and it worked.

v923z, 2023-04-16:
@zachmoshe @hamza-712 If either of you could reproduce these results, it would be a worthwhile addition to https://github.com/v923z/micropython-builder.

zachmoshe, 2023-04-17:
@v923z - I believe this can be closed as we both were able to build the firmware and it probably wasn't a ulab problem anyway ( I guess a WONT_FIX?)

v923z, 2023-04-17:
You can definitely close it, but I still feel that others could profit from an automated build. Let me know if you're interested.

zachmoshe, 2023-04-17:
sure, I'll take a look and submit a PR

closed 2023-04-17 by zachmoshe