v923z/micropython-ulab #596: ulab for Raspberry Pi Pico W

yamamotomas, 2023-04-16:
I have built a firmware for Raspberry Pi Pico using [`rp2.sh`](https://github.com/v923z/micropython-ulab/blob/master/build/rp2.sh).

When I uploaded the firmware to Pico W, it make errors in connecting to WiFi:

```
MicroPython v1.19.1-1016-gb525f1c9e on 2023-04-16; Raspberry Pi Pico with RP2040
Type "help()" for more information.

>>> import network
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'network'
```

How can I build a ulab firmware for Pico W ?

v923z, 2023-04-16:
OK, but this has nothing to do with `ulab` as such. What happens, if you try to compile the firmware with the files from https://github.com/micropython/micropython/?

yamamotomas, 2023-04-17:
> OK, but this has nothing to do with ulab as such.

Your are right. Sorry for a trouble.

In the shell script [https://github.com/v923z/micropython-ulab/blob/master/build/rp2.sh](https://github.com/v923z/micropython-ulab/blob/master/build/rp2.sh),
it tries to download from micropython github site.

```
if [ ! -d $MPY_DIR ]; then
    printf "Cloning MicroPython\n"
    git clone https://github.com/micropython/micropython.git micropython
fi
```

If there be some way to let the compiler know that the target is PICO W,
I'd appreciate it very much.

jimmo, 2023-04-17:
> If there be some way to let the compiler know that the target is PICO W,

@yamamotomas add `BOARD=PICO_W` to the last line.

I would do this by adding a `export BOARD=PICO` (which you can change to PICO_W) to the top, then use

`make USER_C_MODULES=$ULAB_DIR/micropython.cmake BOARD=$BOARD`

(While you're at it, change the `rm -r build` to `make BOARD=$BOARD clean` as what's there currently doesn't do anything because there is no `build` directory, only `build-PICO` etc).

yamamotomas, 2023-04-17:
@jimmo Thank you so much. It worked fine.

The resulting shell script is this:
```
#!/bin/bash

export BOARD=PICO_W
export BUILD_DIR=$(pwd)
export MPY_DIR=$BUILD_DIR/micropython
export ULAB_DIR=$BUILD_DIR/../code

if [ ! -d $ULAB_DIR ]; then
    printf "Cloning ulab\n"
    ULAB_DIR=$BUILD_DIR/ulab/code
    git clone https://github.com/v923z/micropython-ulab.git ulab
fi

if [ ! -d $MPY_DIR ]; then
    printf "Cloning MicroPython\n"
    git clone https://github.com/micropython/micropython.git micropython
fi

cd $MPY_DIR
git submodule update --init
cd ./mpy-cross && make # build cross-compiler (required)

cd $MPY_DIR/ports/rp2
make BOARD=$BOARD clean
make USER_C_MODULES=$ULAB_DIR/micropython.cmake BOARD=$BOARD
```

v923z, 2023-04-17:
@yamamotomas As far as I understand, the problem was that the script on https://github.com/v923z/micropython-builder didn't work properly. Would you mind submitting a PR there?

yamamotomas, 2023-04-17:
@v923z Thank you for your comment.

> As far as I understand, the problem was that the script on https://github.com/v923z/micropython-builder didn't work properly. 

It worked perfectly fine for Raspberry Pi Pico.
But the `uf2` file is not good enough for Pico W (WiFi version), and needs extra module like  `network`.

So my wish is that you put above script in the repository, so that a Raspberry Pi Pico W user can make the firmware quickly and easily.

v923z, 2023-04-17:
People, who contribute, get brownie points: https://github.com/v923z/micropython-builder/graphs/contributors ;-)