v923z/micropython-ulab #608: [FEATURE REQUEST] support reshape(-1) for ndarrays

Derfies, 2023-05-03:
Super handy to be able to flatten an array to 1 dimension in vanilla numpy, eg:
```
arr = np.array([[1, 2, 3], [4, 5, 6]])
print(arr.reshape(-1))
-> [1 2 3 4 5 6]
```
whereas this raises
`TypeError: shape must be a tuple`
in ulab.

Using a tuple, ie arr.reshape((-1,)) gives the same issue unfortunately.

Derfies, 2023-05-03:
Ah. After a little more perusing I take it that .flatten is the way to go here?

v923z, 2023-05-03:
If you want to, we could add `.reshape(-1)` as an alias for `.flatten`. Are they really the same?

Derfies, 2023-05-03:
I've used numpy a fair bit but I don't consider myself an authority on the matter. Perhaps leave it as is rather than confused the next boffin who knows the nuance between these two methods!

v923z, 2023-05-06:
One element in the tuple can actually be -1: https://numpy.org/doc/stable/reference/generated/numpy.reshape.html

v923z, 2023-05-06:
@Derfies `numpy` is not quite pedantic, and while the documentation states that the unknown dimension length should be -1, it is actually not checked. In any case, https://github.com/v923z/micropython-ulab/pull/612 implements this feature.

closed 2023-05-06 by v923z