v923z/micropython-ulab #609: [FEATURE REQUEST] New stable release/tag please ?

iabdalkader, 2023-05-05:
Just requesting a new release/tag to track, since the major change in the mp type definition broke the old release 🙏🏼

v923z, 2023-05-07:
Here you go: https://github.com/v923z/micropython-ulab/releases/tag/6.0.12. 
Would you be interested in adding your build script here: https://github.com/v923z/micropython-builder? People would then always have cutting-edge firmware.

closed 2023-05-07 by v923z

iabdalkader, 2023-05-07:
Thank you!

> Would you be interested in adding your build script

 Yes I'll try to add a script to build the latest board.

v923z, 2023-05-07:
> Yes I'll try to add a script to build the latest board.

If you want to add multiple boards, I would be happy to host them, too.