v923z/micropython-ulab #618: [BUG] crash deleting element of ulab array

jepler, 2023-05-17:
**Describe the bug**
version 6.0.12-2D-c

The following python code crashes:
```py
from ulab import numpy as np
n=np.array([0])
del n[0]
```

**To Reproduce**
I run the above code sample on `micropython-2` built to run on a host computer

**Expected behavior**
standard Python numpy throws an exception: `ValueError: cannot delete array elements`

jepler, 2023-05-17:
should also check slice deletion for similar problem

jepler, 2023-05-17:
The site of the crash is
```
0x0000555555584a03 in ndarray_from_mp_obj (obj=0x0, 
    other_type=other_type@entry=0 '\000')
    at /home/jepler/src/ulab/code/ndarray.c:1696
1696	    if(mp_obj_is_int(obj)) {
(gdb) where
#0  0x0000555555584a03 in ndarray_from_mp_obj (obj=0x0, 
    other_type=other_type@entry=0 '\000')
    at /home/jepler/src/ulab/code/ndarray.c:1696
#1  0x0000555555584e66 in ndarray_subscr (self_in=0x7ffff7a4e360, index=0x1, 
    value=<optimized out>) at /home/jepler/src/ulab/code/ndarray.c:1446
#2  0x00005555555b4a2e in mp_obj_subscr (base=0x7ffff7a4e360, index=0x1, 
    value=0x0) at ../../py/obj.c:538
```
A value of `MP_OBJ_NULL` (0 in this build) is used with the subscr method to delete. I think the proper fix for this is in ulab and raises an error when the argument to `ndarray_subscr` is MP_OBJ_NULL.

v923z, 2023-05-17:
@jepler Thanks for raising the issue and the suggestion! 

Fixed by https://github.com/v923z/micropython-ulab/pull/619.

closed 2023-05-17 by v923z

jepler, 2023-05-17:
Thanks for the fix