v923z/micropython-ulab #622: [BUG]

alibukharai, 2023-05-30:
**Describe the bug**
Hi. Ubuntu version Ubuntu 22.04.2 LTS. I am using ESP-32 S3 (**GENERIC_S3_SPIRAM_OCT**) with ESP-IDF 4.4 (**ESP-IDF v4.4.4-448-gfd5e03b221**) commit id (**commit ed7a3b11d9a6c21a964d55ebfcdefeb392389d10**). when I tryies to install Ulab It gives me an error. Please help me with that. Here is the logs. 

```
(base) ➜  esp32 git:(master) ✗ make         

idf.py -D MICROPY_BOARD=GENERIC_S3_SPIRAM_OCT -D MICROPY_BOARD_DIR=/home/ali/work/learning/Micropython/micropython/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT -B build-GENERIC_S3_SPIRAM_OCT -DUSER_C_MODULES=/ulab/code/micropython.cmake build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)

Executing action: all (aliases: build)

Running cmake in directory /home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT

Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=GENERIC_S3_SPIRAM_OCT -DMICROPY_BOARD_DIR=/home/ali/work/learning/Micropython/micropython/ports/esp32/boards/GENERIC_S3_SPIRAM_OCT -DUSER_C_MODULES=/ulab/code/micropython.cmake -DCCACHE_ENABLE=0 /home/ali/work/learning/Micropython/micropython/ports/esp32"...

-- IDF_TARGET not set, using default target: esp32

-- Found Git: /usr/bin/git (found version "2.34.1") 

-- The C compiler identification is GNU 8.4.0

-- The CXX compiler identification is GNU 8.4.0

-- The ASM compiler identification is GNU

-- Found assembler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /home/ali/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

-- Building ESP-IDF components for target esp32s3

-- Project sdkconfig file /home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig

Loading defaults file /home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined...

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:8 CONFIG_COMPILER_OPTIMIZATION_LEVEL_RELEASE was replaced with CONFIG_COMPILER_OPTIMIZATION_SIZE

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:67 CONFIG_USB_ENABLED was replaced with CONFIG_TINYUSB

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:68 CONFIG_USB_CDC_ENABLED was replaced with CONFIG_TINYUSB_CDC_ENABLED

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:69 CONFIG_USB_CDC_RX_BUFSIZE was replaced with CONFIG_TINYUSB_CDC_RX_BUFSIZE

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:70 CONFIG_USB_CDC_TX_BUFSIZE was replaced with CONFIG_TINYUSB_CDC_TX_BUFSIZE

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:90 CONFIG_DEFAULT_PSRAM_CLK_IO was replaced with CONFIG_SPIRAM_CLK_IO

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:91 CONFIG_DEFAULT_PSRAM_CS_IO was replaced with CONFIG_SPIRAM_CS_IO

/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/sdkconfig.combined:103 CONFIG_FLASHMODE_QIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_QIO

/tmp/confgen_tmpxnu13pb5:74 line was updated to CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=n

[... 45 lines cut ...]

    /ulab/code/micropython.cmake

Call Stack (most recent call first):

  main/CMakeLists.txt:14 (include)

Found User C Module(s): 

-- Found Python3: /usr/bin/python3.10 (found version "3.10.6") found components: Interpreter 

-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32s3 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa

-- Component paths: /home/ali/esp/esp-idf_4.4/components/app_trace /home/ali/esp/esp-idf_4.4/components/app_update /home/ali/esp/esp-idf_4.4/components/asio /home/ali/esp/esp-idf_4.4/components/bootloader /home/ali/esp/esp-idf_4.4/components/bootloader_support /home/ali/esp/esp-idf_4.4/components/bt /home/ali/esp/esp-idf_4.4/components/cbor /home/ali/esp/esp-idf_4.4/components/cmock /home/ali/esp/esp-idf_4.4/components/coap /home/ali/esp/esp-idf_4.4/components/console /home/ali/esp/esp-idf_4.4/components/cxx /home/ali/esp/esp-idf_4.4/components/driver /home/ali/esp/esp-idf_4.4/components/efuse /home/ali/esp/esp-idf_4.4/components/esp-tls /home/ali/esp/esp-idf_4.4/components/esp32s3 /home/ali/esp/esp-idf_4.4/components/esp_adc_cal /home/ali/esp/esp-idf_4.4/components/esp_common /home/ali/esp/esp-idf_4.4/components/esp_eth /home/ali/esp/esp-idf_4.4/components/esp_event /home/ali/esp/esp-idf_4.4/components/esp_gdbstub /home/ali/esp/esp-idf_4.4/components/esp_hid /home/ali/esp/esp-idf_4.4/components/esp_http_client /home/ali/esp/esp-idf_4.4/components/esp_http_server /home/ali/esp/esp-idf_4.4/components/esp_https_ota /home/ali/esp/esp-idf_4.4/components/esp_https_server /home/ali/esp/esp-idf_4.4/components/esp_hw_support /home/ali/esp/esp-idf_4.4/components/esp_ipc /home/ali/esp/esp-idf_4.4/components/esp_lcd /home/ali/esp/esp-idf_4.4/components/esp_local_ctrl /home/ali/esp/esp-idf_4.4/components/esp_netif /home/ali/esp/esp-idf_4.4/components/esp_phy /home/ali/esp/esp-idf_4.4/components/esp_pm /home/ali/esp/esp-idf_4.4/components/esp_ringbuf /home/ali/esp/esp-idf_4.4/components/esp_rom /home/ali/esp/esp-idf_4.4/components/esp_serial_slave_link /home/ali/esp/esp-idf_4.4/components/esp_system /home/ali/esp/esp-idf_4.4/components/esp_timer /home/ali/esp/esp-idf_4.4/components/esp_websocket_client /home/ali/esp/esp-idf_4.4/components/esp_wifi /home/ali/esp/esp-idf_4.4/components/espcoredump /home/ali/esp/esp-idf_4.4/components/esptool_py /home/ali/esp/esp-idf_4.4/components/expat /home/ali/esp/esp-idf_4.4/components/fatfs /home/ali/esp/esp-idf_4.4/components/freemodbus /home/ali/esp/esp-idf_4.4/components/freertos /home/ali/esp/esp-idf_4.4/components/hal /home/ali/esp/esp-idf_4.4/components/heap /home/ali/esp/esp-idf_4.4/components/idf_test /home/ali/esp/esp-idf_4.4/components/ieee802154 /home/ali/esp/esp-idf_4.4/components/jsmn /home/ali/esp/esp-idf_4.4/components/json /home/ali/esp/esp-idf_4.4/components/libsodium /home/ali/esp/esp-idf_4.4/components/log /home/ali/esp/esp-idf_4.4/components/lwip /home/ali/work/learning/Micropython/micropython/ports/esp32/main /home/ali/esp/esp-idf_4.4/components/mbedtls /home/ali/esp/esp-idf_4.4/components/mdns /home/ali/esp/esp-idf_4.4/components/mqtt /home/ali/esp/esp-idf_4.4/components/newlib /home/ali/esp/esp-idf_4.4/components/nghttp /home/ali/esp/esp-idf_4.4/components/nvs_flash /home/ali/esp/esp-idf_4.4/components/openssl /home/ali/esp/esp-idf_4.4/components/openthread /home/ali/esp/esp-idf_4.4/components/partition_table /home/ali/esp/esp-idf_4.4/components/perfmon /home/ali/esp/esp-idf_4.4/components/protobuf-c /home/ali/esp/esp-idf_4.4/components/protocomm /home/ali/esp/esp-idf_4.4/components/pthread /home/ali/esp/esp-idf_4.4/components/sdmmc /home/ali/esp/esp-idf_4.4/components/soc /home/ali/esp/esp-idf_4.4/components/spi_flash /home/ali/esp/esp-idf_4.4/components/spiffs /home/ [... cut ...] components/wear_levelling /home/ali/esp/esp-idf_4.4/components/wifi_provisioning /home/ali/esp/esp-idf_4.4/components/wpa_supplicant /home/ali/esp/esp-idf_4.4/components/xtensa

-- Configuring incomplete, errors occurred!

See also "/home/ali/work/learning/Micropython/micropython/ports/esp32/build-GENERIC_S3_SPIRAM_OCT/CMakeFiles/CMakeOutput.log".

cmake failed with exit code 1

-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting

make: *** [Makefile:52: all] Error 1
```

v923z, 2023-05-30:
You can download the latest `micropthon`/`ulab` firmware from https://github.com/v923z/micropython-builder/releases.

v923z, 2023-05-30:
I think `micropython` should be inside the `ulab` directory. I have the feeling that in your case, it is the other way around. If you clone this repository, then you simply have to run the script that you find in https://github.com/v923z/micropython-ulab/tree/master/build.

v923z, 2023-06-21:
Since this is not really a `ulab` issue, closing now.

closed 2023-06-21 by v923z