v923z/micropython-ulab #634: [BUG]

Derfies, 2023-07-10:
Latest PICO_W firmware build (Release date 2023-07-09) appears to cause the device to not be recognisable on windows.

Steps to reproduce:
- Download [PICO_W.uf2](https://github.com/v923z/micropython-builder/releases/download/latest/PICO_W.uf2)
- Hold boot sel down and plug usb cable into PC
- Copy the firmware over

Expected result:
- Explorer window should disappear and the device should appear in the list of devices Thonny can connect to.

Observed result:
- The explorer window should disappear and window will alert with "USB device not recognised. The last USB device you connected to this computer malfunctioned, and windows does not recognise it"
- The device does not appear in the list of devices Thonny can connect to.

Using the latest [PICO.uf2](https://github.com/v923z/micropython-builder/releases/download/latest/PICO.uf2) firmware instead appears to restore the device to working order, as does installing the latest vanilla micropython [firmware](https://micropython.org/resources/firmware/rp2-pico-20230426-v1.20.0.uf2).

v923z, 2023-07-31:
@Derfies Does the issue still persist?

Derfies, 2024-02-15:
Sorry for leaving this open for so long. No, the issue seems to have been resolved with the latest builds.

closed 2024-02-15 by Derfies

v923z, 2024-02-16:
Thanks for the feedback!