v923z/micropython-ulab #640: [BUG] Not building with latest micropython repo

darshansm, 2023-07-30:
Not building with latest micropython with esp32

jimmo, 2023-07-31:
MicroPython recently moved to using IDF 5.0.2 for ESP32 builds. This is now the only supported version.

See https://github.com/micropython/micropython/blob/master/ports/esp32/README.md#setting-up-esp-idf-and-the-build-environment for build instructions.

The ulab instructions at https://github.com/v923z/micropython-ulab#compiling-with-cmake and https://github.com/v923z/micropython-ulab/blob/master/build/esp32-cmake.sh will need to be updated.

v923z, 2023-07-31:
Thanks, Jim!

v923z, 2023-07-31:
@darshansm 

> Not building with latest micropython with esp32

In the future, please, submit a more informative bug report! It's no accident that there is a template for bugs. A terse statement amounting to something's not working is not going to be enough to find out what the root cause might be. We don't even know which operating system you had, when attempting to compile the firmware. We don't know at which step the build failed, let alone with what error message.

You're more probable to get a prompt response or a speedy fix for your problem, if you share as much information as possible. 

Also, if @jimmo's suggestion helped you, please, consider contributing, and creating a PR with your modifications!

darshansm, 2023-08-01:
@jimmo thanks you for the info. In updated esp32cmake.sh with idf 5.0.2 it is working and able to build