v923z/micropython-ulab #643: [BUG] numpy.where have a strange behavior when receiving np.bool as its conditions

qqice, 2023-08-08:
**Describe the bug**
I use the version 6.4.0-2D-c of ulab. When I use numpy.where(condition,x,y) and pass a variable of type np.bool to the condition parameter, the code exhibits different behavior than standard numpy.

**To Reproduce**
```python
from ulab import numpy as np

BLACK = np.array([0, 0, 0],dtype=np.uint8)
BLUE = np.array([0, 0, 255],dtype=np.uint8)

d = np.array([False,True,True], dtype=np.bool)
f = np.array([0,1,1], dtype=np.uint8)

print(np.where(d[1], BLUE, BLACK))
print(np.where(f[1], BLUE, BLACK))
```

**Expected behavior**
the program should print
```python
array([0, 0, 255], dtype=uint8)
array([0, 0, 255], dtype=uint8)
```
instead of actrually printing
```python
array([0, 0, 0], dtype=uint8)
array([0, 0, 255], dtype=uint8)
```
as the similar code on my computer
```python
import numpy as np

BLACK = np.array([0, 0, 0],dtype=np.uint8)
BLUE = np.array([0, 0, 255],dtype=np.uint8)

d = np.array([False,True,True], dtype=np.bool_)
f = np.array([0,1,1], dtype=np.uint8)

print(np.where(d[1], BLUE, BLACK))
print(np.where(f[1], BLUE, BLACK))
```
prints
```python
[  0   0 255]
[  0   0 255]
```

**Additional context**
You may notice that the code on my PC uses dtype=np.bool_ instead of dtype=np.bool. That's because when I use np.bool on my PC, it threw out an exception
```
C:\Users\qqice\PycharmProjects\pythonProject1\main.py:6: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  d = np.array([False,True,True], dtype=np.bool)
Traceback (most recent call last):
  File "C:\Users\qqice\PycharmProjects\pythonProject1\main.py", line 6, in <module>
    d = np.array([False,True,True], dtype=np.bool)
  File "C:\Users\qqice\PycharmProjects\pythonProject1\venv\lib\site-packages\numpy\__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'bool'.
`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations. Did you mean: 'bool_'?
```

v923z, 2023-08-08:
I'm not sure I understand what the issue is. Here is my test file

```python
from ulab import numpy as np
import ulab

print(ulab.__version__)

BLACK = np.array([0, 0, 0],dtype=np.uint8)
BLUE = np.array([0, 0, 255],dtype=np.uint8)

d = np.array([False,True,True], dtype=np.bool)
f = np.array([0,1,1], dtype=np.uint8)

print('test with d: ', np.where(d, BLUE, BLACK))
print('test with f: ', np.where(f, BLUE, BLACK))

print('test with d[1]: ', np.where(d[1], BLUE, BLACK))
print('test with f[1]: ', np.where(f[1], BLUE, BLACK))
```
and here is the output

```python
6.4.0-2D-c
test with d:  array([0, 0, 255], dtype=uint8)
test with f:  array([0, 0, 255], dtype=uint8)
test with d[1]:  array([0, 0, 255], dtype=uint8)
test with f[1]:  array([0, 0, 255], dtype=uint8)
```

Is this not what you expect? Your condition is not the full array, but only the second element, which is true (1) in both cases, so the result is the complete BLUE array.

qqice, 2023-08-09:
> 

Yes, but in my firmware, your code prints the following output:
```python
6.4.0-2D-c
test with d:  array([0, 0, 255], dtype=uint8)
test with f:  array([0, 0, 255], dtype=uint8)
test with d[1]:  array([0, 0, 0], dtype=uint8)
test with f[1]:  array([0, 0, 255], dtype=uint8)
```
Do I need to attach my firmware here for further analysis?

v923z, 2023-08-09:
Can you try the firmware from here https://github.com/v923z/micropython-builder?

qqice, 2023-08-09:
I've tried the firmware [PICO.uf2](https://github.com/v923z/micropython-builder/releases/download/latest/PICO.uf2) from [https://github.com/v923z/micropython-builder](https://github.com/v923z/micropython-builder). However, your code also prints
```python
6.4.0-2D-c
test with d:  array([0, 0, 255], dtype=uint8)
test with f:  array([0, 0, 255], dtype=uint8)
test with d[1]:  array([0, 0, 0], dtype=uint8)
test with f[1]:  array([0, 0, 255], dtype=uint8)
```

v923z, 2023-08-09:
OK, then there seems to be some discrepancy between the `unix` and the `rp2040`. 

@jimmo @dpgeorge can you comment on this? Do you think that the issue is in `micropython` itself?

The problem pops up, when https://github.com/v923z/micropython-ulab/blob/5279de73abddf75f702622661346a79f39896aea/code/numpy/compare.c#L513 is called. That function is assigned in https://github.com/v923z/micropython-ulab/blob/5279de73abddf75f702622661346a79f39896aea/code/numpy/compare.c#L488, and is implemented in https://github.com/v923z/micropython-ulab/blob/5279de73abddf75f702622661346a79f39896aea/code/ulab_tools.c#L49.

dpgeorge, 2023-08-10:
This should fix it:
```diff
--- a/code/ndarray.c
+++ b/code/ndarray.c
@@ -1738,7 +1738,7 @@ ndarray_obj_t *ndarray_from_mp_obj(mp_obj_t obj, uint8_t other_type) {
         mp_float_t *array = (mp_float_t *)ndarray->array;
         array[0] = mp_obj_get_float(obj);
     } else if(mp_obj_is_bool(obj)) {
-        ndarray = ndarray_new_linear_array(1, NDARRAY_BOOLEAN);
+        ndarray = ndarray_new_linear_array(1, NDARRAY_BOOL);
         uint8_t *array = (uint8_t *)ndarray->array;
         if(obj == mp_const_true) {
             *array = 1;
```
I suggest to make a test for this case, and also change the type of NDARRAY_BOOLEAN so it's incompatible with NDARRAY_BOOL (eg make it a pointer to something).  Also might be worth renaming NDARRAY_BOOLEAN to make it clear where/when it can be used.

v923z, 2023-08-10:
> This should fix it:

Gosh, this was a really stupid mistake! Thanks for digging it out!

> I suggest to make a test for this case, and also change the type of NDARRAY_BOOLEAN so it's incompatible with NDARRAY_BOOL (eg make it a pointer to something). Also might be worth renaming NDARRAY_BOOLEAN to make it clear where/when it can be used.

OK, good point. I still wonder, though, why the `unix` port didn't show this behaviour. The only difference to the `rp2040` port should be the size of `float`, nothing more. Or is there something else at play here?

v923z, 2023-08-10:
@qqice Can you patch up your local copy, and confirm the fix? Thereafter, could you, please, create a PR here? I can guide you through the steps, should you need help.

dpgeorge, 2023-08-10:
> The only difference to the `rp2040` port should be the size of `float`, nothing more.

It was accessing the uint8 memory as a float/double, so getting the wrong value.  On unix it was a fluke that the value just happened to be correct!

qqice, 2023-08-10:
> @qqice Can you patch up your local copy, and confirm the fix?

Sure, with great pleasure. After the patch, the code prints out the following output
```python
6.4.0-2D-c
test with d:  array([0, 0, 255], dtype=uint8)
test with f:  array([0, 0, 255], dtype=uint8)
test with d[1]:  array([0, 0, 255], dtype=uint8)
test with f[1]:  array([0, 0, 255], dtype=uint8)
```
which is in line with the performance of numpy on a computer.
I‘ve created a PR for the issue. Please check and see what else I can do for you.

v923z, 2023-08-10:
@qqice Thanks! Could you, please, increment the version number in https://github.com/v923z/micropython-ulab/blob/5279de73abddf75f702622661346a79f39896aea/code/ulab.c#L36 to 6.4.1, and add a short note in the https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-change-log.md?

v923z, 2023-08-10:
> > The only difference to the `rp2040` port should be the size of `float`, nothing more.
> 
> It was accessing the uint8 memory as a float/double, so getting the wrong value. On unix it was a fluke that the value just happened to be correct!

OK, that makes sense. Again, thanks for your comments!

qqice, 2023-08-10:
> @qqice Thanks! Could you, please, increment the version number in
> 
> https://github.com/v923z/micropython-ulab/blob/5279de73abddf75f702622661346a79f39896aea/code/ulab.c#L36
> 
> to 6.4.1, and add a short note in the https://github.com/v923z/micropython-ulab/blob/master/docs/ulab-change-log.md?

Done. Please check.

v923z, 2023-08-10:
@qqice Many thanks for contributing!

commit a05ec053 (2023-08-10): Fix #643 (#645)

* Fix #643

* Update to version 6.4.1