v923z/micropython-ulab #646: [BUG] IndexError: indices must be integers, slices, or Boolean lists

raghunila, 2023-09-13:
**Describe the bug**
built from main branch
Issue is while rearranging 2 dimensional array, giving below error
IndexError: indices must be integers, slices, or Boolean lists
but works good with python 3.2 and numpy package

```python
from ulab import numpy as np
from time import sleep

res = []
for i in range(10):
    res.append(np.arange(10 * i, 10 * (i + 1)))
res = np.array(res,  dtype=np.uint8)
res1 = np.array([1, 2, 3, 4, 5, 6, 7, 8, 9, 0],  dtype=np.uint8)
print(res)

print(res[:, res1])

print(ulab.__version__)
```

**To Reproduce**
run this python console with ulab build

**Expected behavior**
columns should re arrange.

**Additional context**

v923z, 2023-09-14:
I don't see what the problem is. You're trying to index by an `ndarray`. The `IndexError` tells you that that is not an accepted type. Implementing all aspects of `numpy` indexing is highly non-trivial, consumes a lot of flash space, and RAM. For that reason, you can't do everything that you can do in `numpy`.

v923z, 2024-10-07:
I think https://github.com/v923z/micropython-ulab/tree/take addresses the issue raised here, so I'll close this, one `take` is merged into `master`.

closed 2024-10-09 by v923z