v923z/micropython-ulab #656: [BUG]

ValergooD, 2024-01-15:
**Describe the bug**
A clear and concise description of what the bug is. Give the `ulab` version
Error message after build 
Makefile:18: The PICO* boards have been renamed to RPI_PICO* Makefile:20: *** Invalid BOARD specified: boards/PICO_W.  Stop. Makefile:18: The PICO* boards have been renamed to RPI_PICO* Makefile:20: *** Invalid BOARD specified: boards/PICO_W.  Stop.
For fix need use this in file "rp2w.sh" change "BOARD=PICO_W" to  "BOARD=RPI_PICO_W"
```python
import ulab
print(ulab.__version__)
```

**To Reproduce**
Describe the steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context that might help to locate the root of the problem.

v923z, 2024-01-16:
Thanks for the bug report!

v923z, 2024-01-16:
Fixed, closing now.

closed 2024-01-16 by v923z