v923z/micropython-ulab #658: [BUG] Reshaping a slice creates an empty array

furbrain, 2024-02-07:
**Describe the bug**
Calling ``reshape`` on a slice of an array can lead to an empty array.
ulab version 6.4.1-2D

**To Reproduce**
```python
>>> a = np.array(range(36)).reshape((6,6))
>>> a
array([[0.0, 1.0, 2.0, 3.0, 4.0, 5.0],
       [6.0, 7.0, 8.0, 9.0, 10.0, 11.0],
       [12.0, 13.0, 14.0, 15.0, 16.0, 17.0],
       [18.0, 19.0, 20.0, 21.0, 22.0, 23.0],
       [24.0, 25.0, 26.0, 27.0, 28.0, 29.0],
       [30.0, 31.0, 32.0, 33.0, 34.0, 35.0]], dtype=float32)
>>> b = a[0,:]
>>> b.reshape((1,6))
array([[0.0, 1.0, 2.0, 3.0, 4.0, 5.0]], dtype=float32)
>>> c = a[:,0]
>>> c.reshape((1,6))
array([], shape=(0,6), dtype=float32)
```

**Expected behavior**

``c`` should be ``[[0.0, 6.0, 12.0, 18.0, 24.0, 30.0]]``, not ``[]``

**Additional context**

This seems to be a result of recent changes. Ulab (6.0.12) that came with circuitpython 8.1 did not have this issue

v923z, 2024-02-07:
I get 

```python
c.reshape((1,6))
array([[0.0],
       [6.0],
       [12.0],
       [18.0],
       [24.0],
       [30.0]], dtype=float64)
```
which is also wrong.

furbrain, 2024-02-08:
I think that is correct: c is looking at the first column, b is looking at the first row. My demo code was running on a nrf52840 in circuitpython

Phil

furbrain, 2024-02-08:
Sorry, didn't look at your result properly. I see they issue...

furbrain, 2024-02-08:
Sorry, didn't look at your result properly. I see they issue...

Just had a look at numpy docs for reshape: "it is not always possible to change the shape of an array without copying the data"

v923z, 2024-02-10:
Yes, that's correct, but your particular example should still be doable without copying. I'll try to sort out the issue.

v923z, 2024-02-10:
Fixed in https://github.com/v923z/micropython-ulab/pull/660

closed 2024-02-10 by v923z