v923z/micropython-ulab #661: [FEATURE REQUEST] implement numpy.take

v923z, 2024-02-16:
**Describe the solution you'd like**
Implement fancy indexing based on https://numpy.org/doc/stable/reference/generated/numpy.take.html 

**Additional context**
The implementation would resolve https://github.com/v923z/micropython-ulab/issues/607 in the most frequent use cases.

v923z, 2024-03-26:
Code added in https://github.com/v923z/micropython-ulab/tree/take

Derfies, 2024-07-22:
@v923z sorry to have dropped off. I will pick this up for testing.

Derfies, 2024-07-26:
Hi @v923z just built this on windows using Ubuntu and it seems to have produced the firmware as expected. Running the following snippet however:
```
a = np.ndarray([1, 2, 3], dtype=np.uint8)
indices = np.ndarray([0], dtype=np.uint8)
print(np.take(a, indices))
```
...produces None in the console. Seems like I can use the 'out' kwarg to get a result, so I tried that too:
```
a = np.ndarray([1, 2, 3], dtype=np.uint8)
indices = np.ndarray([1], dtype=np.uint8)
out = np.empty(indices.shape, dtype=np.uint8)
np.take(a, indices, out=out)
print(out)
```
...produces array([0], dtype=uint8) where I would expect the value array([2], dtype=uint8). Changing the 'indices' array to take more values does indeed give me more values in the out array, however they always appear to be 0.

I'm no C++ dev by a long shot, but I can see the only return statement in create_take returns mp_const_none, so that's my guess for the first issue. Will keep reading this PR to see if there's an obvious reason for returning zeros.

Derfies, 2024-07-27:
Ok had a bit of a fiddle. It seems like we create the output array using ndarray_new_dense_ndarray but don't assign the elements to it. I tried:
```
for(size_t i=0; i < indices_len; i++) {
    ndarray_set_value(a->dtype, out->array, i, mp_obj_new_int(indices[i]));
```
Which didn't give me the results I hoped for, but I have to admit I'm stumbling in the dark at the moment!

Also just wondering is https://github.com/v923z/micropython-ulab/pull/651/files comparable functionality?

v923z, 2024-07-27:
Thanks for taking a look at it! I know that there is a glitch in the implementation. I'll try to fix it in the coming days.

v923z, 2024-10-05:
@Derfies Can you check out the code again? The test cases pass.

v923z, 2024-10-09:
Implemented in https://github.com/v923z/micropython-ulab/pull/688.

closed 2024-10-09 by v923z