v923z/micropython-ulab #662: [BUG] Some type annotations don't include scalars

kbsriram, 2024-02-20:
**Describe the bug**
Ulab version: `6.5.1-2D-c`

Type annotations on some functions don't appear to include scalars.

**To Reproduce**

For instance, this code:

```python
from ulab import numpy as np

np.acos(0)
```
when used with the [current type annotation](https://github.com/v923z/micropython-ulab/blob/acfec3e9af5d351df88a7d71786f11a12f7545d3/code/numpy/vector.c#L237-L243) extracted by circuitpython produces this error with `mypy`.
```shell
$ export MYPYPATH=${HOME}/circuitpython/circuitpython-stubs
$ mypy --disallow-untyped-defs type_test.py
type_test.py:3: error: Argument 1 to "acos" has incompatible type "int"; expected "Union[ndarray, List[float], Tuple[float], range]"  [arg-type]
Found 1 error in 1 file (checked 1 source file)
```

**Expected behavior**
The code accepts both scalars and `_ArrayLike` - so I was expecting type-checking to work.

**Additional context**
Would one option be to add a TypeAlias that unions scalars with _ArrayLike, and use that for methods that accept or produce both scalars and arrays?

v923z, 2024-02-20:
The issue concerns the documentation and not the code directly, right?

v923z, 2024-02-20:
OK. Then it's not a bug.

closed 2024-02-28 by v923z