v923z/micropython-ulab #671: [FEATURE REQUEST]

kwagyeman, 2024-07-09:
**Describe the solution you'd like**
I would like if append() and vstack() could be implemented.

Also if this limitation could be removed.

> Indices can also be a list of Booleans. By using a Boolean list, we can select those elements of an array that satisfy a specific condition. At the moment, such indexing is defined for row vectors only; when the rank of the tensor is higher than 1, the function raises a NotImplementedError exception, though this will be rectified in a future version of ulab.

**Additional context**
I was trying to implement non-max-suppression using Ulab and ran into the lack of these features complicating things for me.

Happy to finance this work.

v923z, 2024-07-09:
[append](https://numpy.org/doc/stable/reference/generated/numpy.append.html) should be relatively straightforward, and others might also find it useful, so I would be happy to add it.

Reading the documentation of [vstack](https://numpy.org/doc/stable/reference/generated/numpy.vstack.html), it seems to me that it's only a shortcut to what you can already do: [concatenate](https://micropython-ulab.readthedocs.io/en/latest/ulab-ndarray.html#concatenate) has already been implemented. If that doesn't match your requirements, we could discuss the addition of `vstack`. 

> Also if this limitation could be removed.

Do you mean the fact that Boolean indices are supposed to be 1D arrays, or something else?

kwagyeman, 2024-07-09:
> Reading the documentation of [vstack](https://numpy.org/doc/stable/reference/generated/numpy.vstack.html), it seems to me that it's only a shortcut to what you can already do: [...]

It's more about making it easier to use desktop code directly without having to modify it too much. vstack would just be a wrapper around concatenate.

> Do you mean the fact that Boolean indices are supposed to be 1D arrays, or something else?

If I try to use a 1D array of booleans to select rows from a 2D array you get a crash right now. Not a Not implemented error.

`np_boxes` is a 2D array of (x, y, w, h, score, class).

```
filtered_boxes = np_boxes[np_boxes[:, 4] > 0.0]
```

v923z, 2024-07-09:
> If I try to use a 1D array of booleans to select rows from a 2D array you get a crash right now. Not a Not implemented error.
> 
> `np_boxes` is a 2D array of (x, y, w, h, score, class).
> 
> ```
> filtered_boxes = np_boxes[np_boxes[:, 4] > 0.0]
> ```
> 
> The above crashes.

OK, this is a different issue then, thanks for bringing it up! I opened a ticket here https://github.com/v923z/micropython-ulab/issues/672, so that it can be addressed.

kwagyeman, 2024-10-02:
@v923z - Any updates on this?

v923z, 2024-10-07:
Merging the implementation of `take` would address the issue raised here, too, right?

kwagyeman, 2024-10-07:
This is more of just a wrapper around concatenate so that append/vstack work. Would be great to have them as normal so desktop code just works.

v923z, 2024-10-07:
Wrapper should/could be added in https://github.com/v923z/micropython-ulab/tree/master/snippets. That has the advantage that that doesn't consume irrecoverable flash space.

kwagyeman, 2024-10-07:
If it's controlled by an ifdef that works for me. It can be off by default and I can enable it for the OpenMV Cams.

v923z, 2024-10-07:
Everything in https://github.com/v923z/micropython-ulab/tree/master/snippets is in the `python` layer, so you can simply decide, whether you ship the wrapper with your firmware.

v923z, 2024-10-09:
Addressed by https://github.com/v923z/micropython-ulab/pull/688.

closed 2024-10-09 by v923z