v923z/micropython-ulab #673: shape order

iabdalkader, 2024-07-12:
Hi!! Not really a bug but I was wondering if storing the array's shape in reverse is by design ?

For example, with 1x36 array, NDMIS=4 the shape looks like this:

```C
$23 = {base = {type = 0x80195330 <ulab_ndarray_type>}, dtype = 98 'b', itemsize = 1 '\001', boolean = 0 '\000', 
  ndim = 2 '\002', len = 36, shape = {0, 0, 1, 36}, strides = {0, 0, 36, 1}, array = 0x80098a0, origin = 0x80098a0}
```

Naturally I expected the shape to be `{1, 36, 0, 0}`. Is there an option to reverse it back ?

```python
import ulab
print(ulab.__version__)

6.5.2-(3)D
```

**To Reproduce**
Build with NDIMS > Create an array and print shape.

v923z, 2024-07-13:
This is the convention adopted by `numpy`:

```python
>>> import numpy as np
>>> np.ones((1, 36), dtype=np.uint8).strides
(36, 1)
>>> np.ones((1, 36), dtype=np.uint8).shape
(1, 36)
```
and `ulab` follows that:

```python
>>> from ulab import numpy as np
>>> np.ones((1, 36), dtype=np.uint8).strides
(36, 1)
>>> np.ones((1, 36), dtype=np.uint8).shape
(1, 36)
```

The reason is that it's much easier to deal with the nested loops, if the last dimension of the array (the one that has the smallest stride) is in the innermost loop. This is especially true in the case of `ulab`, where, for the sake of performance, storage etc., the dimension is fixed at compile time.

iabdalkader, 2024-07-14:
Ah I see. Thanks for the clarification! I've updated my code to something like this:
```C
for (size_t i = 0; i < input_array->ndim; i++) {
    size_t ulab_offset = ULAB_MAX_DIMS - input_array->ndim;
    if (input_array->shape[ulab_offset + i] != mp_obj_get_int(input_shape->items[i])) {
```

closed 2024-07-14 by iabdalkader