v923z/micropython-ulab #679: [BUG]

OrenFivel, 2024-08-08:
**Describe the bug**
A clear and concise description of what the bug is. Give the `ulab` version
Hi Zoltán Vörös,
My name is Oren Fivel, I try to install ulab into my Raspberry Py PICO controller device, but  I failed.
I installed RPI_PICO.uf2 file into my PICO device using BOOTSEL push-button,
i can import machine  but i can't import ulab.
what i do wrong?
please advise,
Oren.

```python
import ulab
print(ulab.__version__)
```

**To Reproduce**
Describe the steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context that might help to locate the root of the problem.

v923z, 2024-08-10:
It's a bit hard to comment on your problem, for you don't tell where the firmware is from. If you're working with thonny, you can easily update your firmware through its interface.

v923z, 2024-10-07:
Since there is no activity on this issue, I'm closing it now.