v923z/micropython-ulab #690: [FEATURE REQUEST] Implement `unpackbits()`

page200, 2024-11-06:
Are there plans to implement [`unpackbits()`](https://numpy.org/doc/stable/reference/generated/numpy.unpackbits.html)?

And what are good practices for getting the same result **_fast_** while `unpackbits()` is not implemented? How does NumPy do it? [The NumPy implementation](https://github.com/numpy/numpy/blob/fa111b44cfeea4694b38b48f3dbed871a4becc5c/numpy/_core/multiarray.py#L1299) looks like it does nothing, maybe [the decorator `@array_function_from_c_func_and_dispatcher(_multiarray_umath.unpackbits)` of that function](https://github.com/numpy/numpy/blob/fa111b44cfeea4694b38b48f3dbed871a4becc5c/numpy/_core/multiarray.py#L1222) does some magic, but where is it implemented?

I'd like to use the resulting bits as Boolean indices into a constant integer array `X` or do `np.nditer(np.nonzero(X))` or whatever is fastest for achieving the same effect.

page200, 2024-11-06:
[Here is `unpack_bits()`](https://github.com/numpy/numpy/blob/397b7a85943fbaf9e9f34a11e3ce372059e6b9c7/numpy/_core/src/multiarray/compiled_base.c#L1796) in `numpy/numpy/_core/src/multiarray/compiled_base.c`

v923z, 2024-11-06:
Ha! I actually thought of this a while ago, but then I didn't see any immediate use, so I didn't pursue the idea. If you think it's useful, we could definitely add this.

page200, 2024-11-06:
It would be useful.

Maybe we can even forgo the overhead of the parameter checks that NumPy does, or at least do "memoization" so that those checks are performed only once instead of in each call?