v923z/micropython-ulab #697: [FEATURE REQUEST] Support for keepdims?

kwagyeman, 2024-11-26:
For processing things like yolo in ulab the vector size could be 1:N. In the case of 1 we need support for keepdims to prevent dimensionality reduction from happening: https://github.com/openmv/openmv/pull/2521/files

v923z, 2024-11-26:
There are other instances, where keeping the dimension would be useful. It shouldn't be too hard, I simply didn't have time to implement it.

kwagyeman, 2024-11-26:
Looks like you could just add it as an arg here and handle the return of a scalar from each case and turn that scalar back into an array: https://github.com/v923z/micropython-ulab/blob/master/code/numpy/numerical.c

v923z, 2024-11-26:
That would cover the case, when the return value is a scalar, but you might also want to keep the dimension, when the tensor is contracted. So, it's a bit more involving. I'll take a stab at it over the weekend.

kwagyeman, 2024-11-26:
Thank you! Yes, handling this otherwise gets quite ugly quickly.

closed 2025-01-19 by kwagyeman