v923z/micropython-ulab #700: [BUG] Memory Corruption Cases by ULAB subscript access.

kwagyeman, 2024-12-25:
**Describe the bug**
In ulab 6.7.0, accessing a large numpy array using the [] operator causes memory corruption, leading to a system crash. This has been verified using vanilla micropython 1.24. with ulab added as a user_c_module. 

**To Reproduce**
Please see the attached zip file. It contains an `array.bin` file to write to the uPY flash. Then it has three test scripts. The passing script loads the array.bin file, converts it to an ndarray, and then flattens the array and turns it into a list before processing it. This works.

I do the same thing in the error_case_2 script, but, don't turn the flattened numpy array into a list. Memory corruption then happens which leads to objects on the stack not being marked by gc.collect() before it sweeps causing them to be deallocated while still in use.

Finally, error_case shows off that the error happens even without a flattened numpy array. Pretty much any numpy [] access results in memory corruption happening.

**Expected behavior**
All 3 of the scripts should be able to run without the system crashing.

**Additional context**
Some type of stack corruption is happening, leading to objects currently in scope in MicroPython being marked for deletion by gc.collect() while they are still in use. We debugged that gc.collect() does not mark objects currently in scope before it sweeps, which is what causes the crash. 

The question is why are there objects currently in scope that are not marked when gc.collect() runs. Given the difference between scripts is the use of [] access it must be something related to that.

[ulab_issue.zip](https://github.com/user-attachments/files/18244594/ulab_issue.zip)

kwagyeman, 2024-12-25:
FYI - @iabdalkader

iabdalkader, 2024-12-26:
Hi Zoltan, I've managed to reproduce the issue on stm32 port and a new port in development, and we're also asking for help, but I think the issue is ndarray not in MicroPython in general. I've tried many different things to debug this, but so far I haven't been able to find the exact issue so I can't be more specific sorry.

There's one thing seems to "fix" it: If I increase `MICROPY_BYTES_PER_GC_BLOCK` to 64 bytes, without changing anything else, the code runs fine and GC finds the list reference on the stack every time and marks it, so it doesn't get collected while it's still in use. This is not a real fix of course, but it may confirm that there's memory corruption happening somewhere in ndarray functions, as the bigger block would add some unused padding which can be corrupted without causing issue.

v923z, 2024-12-26:
So, do I understand correctly that the issue might be with how the `ndarray` is created? I don't have an STM32 MCU with me at the moment, but I could order a pyboard from digikey, and that should be here by Monday, then.

v923z, 2024-12-26:
Do you still have the problem, if you simply do 

```python
while True:
    with open("array.bin", "rb") as f:
        print("loop", c, "", end='')
        c += 1
        b = f.read()
        a = np.frombuffer(b, dtype=np.float)
        a = a.reshape((1, 3087, 6))

```
The `yolo_v5_postprocess` class doesn't seem to involve anything `ulab`-related. I'm just trying to find out where the issue stems from.

kwagyeman, 2024-12-26:
@v923z - I'll send you code that you can use to build the firmware for the Alif dev kit and test with.

As for the error, it pretty much happens using using ndarray subscr access. If I remove that from the code then it's fine. I'll bug this again later today.

v923z, 2024-12-26:
I understand that the problem is in `ulab`, I'm not questioning that. All I'm saying is that the small snippet above would allow us to narrow down the number of possible culprits, if that still fails.

iabdalkader, 2024-12-27:
> Do you still have the problem, if you simply do

@v923z No, it runs fine, up to 1000s of loops.

dpgeorge, 2024-12-27:
Is this bug reproducible using stock stm32 hardware (eg a pyboard or Nucleo) and stock MicroPython?  If so can you please specify the hardware and the make command used to build (including options to enable ulab) and then I could see if I can reproduce it.

iabdalkader, 2024-12-27:
No, I'm only able to reproduce it using either the N6 or the AE3. @kwagyeman is testing with the Alif kit now using the latest firmware and will let you know how to reproduce.

iabdalkader, 2024-12-27:
@v923z We're doing some more testing, but it may not be a ulab issue at all, just happened to be triggered by ulab. Sorry it's very hard to be certain at this point.

v923z, 2024-12-27:
@iabdalkader Oh, don't worry at all, it's completely fine.

v923z, 2024-12-27:
@dpgeorge Thanks for chipping in!

kwagyeman, 2024-12-27:
@v923z - I've managed to reliably reproduce the issue without ulab. I'll close this ticket.

closed 2024-12-27 by kwagyeman

v923z, 2024-12-27:
> @v923z - I've managed to reliably reproduce the issue without ulab. I'll close this ticket.

Could you, please, leave a link (or just mention this ticket in whatever `micropython` issue you're going to open)? Many thanks!

kwagyeman, 2024-12-27:
Sure, it's on a private repo right now. When it makes it upstream I will link to it.

kwagyeman, 2024-12-30:
Damien found the issue:

When the GC heap is larger than 1MiB the MICROPY_GC_STACK_ENTRY_TYPE needs to be larger than 16-bits wide.

Also, increase MICROPY_ALLOC_GC_STACK_SIZE to speed up GC collection when there are many objects to collect.

[image]

v923z, 2024-12-30:
OK, thanks!