v923z/micropython-ulab #708: [BUG] Unnecessary Tuple

kwagyeman, 2025-01-27:
For our YOLO processing code we do:

```
# Threshold all the scores
score_indices = colum_outputs[:, _YOLO_V5_SCORE]
score_indices = np.nonzero(score_indices > self.threshold)
if isinstance(score_indices, tuple):
    score_indices = score_indices[0]
if not len(score_indices):
    return _NO_DETECTION
```

However, it should just be:

```
# Threshold all the scores
score_indices = colum_outputs[:, _YOLO_V5_SCORE]
score_indices = np.nonzero(score_indices > self.threshold)
if not len(score_indices):
    return _NO_DETECTION
```

Could that tuple appearing be fixed?

v923z, 2025-01-27:
Do you mean in nonzero?

v923z, 2025-01-27:
It seems to me that nonzero returns tuples: https://numpy.org/doc/stable/reference/generated/numpy.nonzero.html

kwagyeman, 2025-01-27:
Yes, you are right, this behavior is correct and matches numpy. ChatGPT was giving me bad code :/

closed 2025-01-27 by kwagyeman