v923z/micropython-ulab #709: [BUG] loadtxt is sensitive to blank lines

Bzero, 2025-01-29:
**Describe the bug**
[loadtxt](https://micropython-ulab.readthedocs.io/en/latest/numpy-functions.html#loadtxt) infers a wrong shape and omits values if the datafile contains blank lines. 

ulab version: 6.7.2-2D

**To Reproduce**
Create a data file with either of the following content (note the white space):

```
0 1

2 3
```

or 

```
0 1
2 3

```

And try to load either of the files with loadtxt:

```
from ulab import numpy
print(numpy.loadtxt('data.dat'))
```

Both result in an array with too few columns and an omitted value:

```
array([[0.0],
       [1.0],
       [2.0]], dtype=float32)
```

**Expected behavior**

Blank lines should not make a difference.

**Additional context**

If the file contains more blank lines `0.0` is inserted, e.g. the file 

```
0 1
2 3

```

results in the array

```
array([[0.0],
       [1.0],
       [2.0],
       [3.0],
       [0.0],
       [0.0],
       [0.0]], dtype=float32)
```

This issue may be related to #706.

v923z, 2025-02-01:
Yes, it's basically the same issue as https://github.com/v923z/micropython-ulab/issues/706.

Bzero, 2025-06-29:
Generally I am concerned about any lines containing only whitespace. I agree that there are multiple ways a blank line in a data file could be interpreted, however I would still hold that omitting elements or adding zeros is not one of the ways that make much sense when it comes to handle blank lines. Of course one can just consider it an invalid datafile but for compatibility with numpy loadtxt I would still suggest to simply omit blank lines.