v923z/micropython-ulab #710: [BUG] Multiline header in savetxt

Bzero, 2025-02-11:
**Describe the bug**
When passing a header containing a linebreak to [savetxt](https://micropython-ulab.readthedocs.io/en/latest/numpy-functions.html#savetxt) only the first line is a comment line, all following lines are uncommented.

ulab version: 6.7.2-2D

**To Reproduce**
```
np.savetxt('loadtxt.dat', np.array([1,2,3]), header='A header\nwith a second line')
```

The second line of the header is not a comment line:
```
# A header
with a second line
1.000000000000000
2.000000000000000
3.000000000000000
```

**Expected behavior**

The second line of the header should be a comment line as well as it is the case with numpy:
```
# A header
# with a second line
1.000000000000000
2.000000000000000
3.000000000000000
```

**Additional context**
Relevant for footers too.

v923z, 2025-02-11:
This is really along the same lines as https://github.com/v923z/micropython-ulab/issues/709, thanks for pointing it out! I'll try to set a bit of time aside to fix this.

v923z, 2025-02-13:
Oh, don't worry! While these are similar issues, it's good that you've opened two tickets.

closed 2025-06-06 by v923z

Bzero, 2025-06-29:
Thank you very much.