v923z/micropython-ulab #713: [BUG]"EPS_5" redefined

Ansen88, 2025-03-14:
**Describe the bug**
A clear and concise description of what the bug is. Give the `ulab` version

The verison is 6.7.3-2D-c
esp-idf v5.4

When compiling,there is a warning:
lab/code/scipy/integrate/integrate.c:58:9: warning: "EPS_5" redefined
58 | #define EPS_5          MICROPY_FLOAT_CONST(1e-5)

esp/v5.4/esp-idf/components/xtensa/esp32s3/include/xtensa/config/specreg.h:72:9: note: this is the location of the previous definition
   72 | #define EPS_5           197

**To Reproduce**
Compiling

**Expected behavior**
In integrate.c, the EPS_5 is 1e-5

**Additional context**
Rename EPS_5 to EPS_51 in integrate.c.

v923z, 2025-03-16:
Thanks for pointing out the issue!

v923z, 2025-03-16:
Merged fix into master, closing it now.

closed 2025-03-16 by v923z