v923z/micropython-ulab #731: [FEATURE REQUEST] `axes` argument for `ndarray.transpose()`

sfe-SparkFro, 2025-07-22:
**Describe the solution you'd like**

In standard NumPy, [`ndarray.transpose()`](https://numpy.org/doc/stable/reference/generated/numpy.ndarray.transpose.html) accepts an optional `axes` argument. For example

```
import numpy as np

my_array = np.zeros((1,2,3))
transposed_array = my_array.transpose((2, 0, 1))

print("Original array shape:", my_array.shape)
print("Transposed array shape:", transposed_array.shape)
```

Prints:

```
Original array shape: (1, 2, 3)
Transposed array shape: (3, 1, 2)
```

sfe-SparkFro, 2025-07-23:
I'm using an array to represent an image, so it needs 3 dimensions (rows, columns, channels) in that order. I need to rotate the image in increments of 90 degrees. The fastest way to do that is to transpose the first 2 axes (columns, rows, channels) then flip one of those first 2 axes (just a change of the strides and shape).

Without being able to specify the axes, transposing results in (channels, columns, rows), which is not what's needed. I'm open to alternative solutions, as long as performance is good with minimal code (eg. can't copy the channels one by one).

v923z, 2025-07-23:
Can you post a `python` snippet that would do what you want? I don't mean `numpy` code, I mean raw `python`. Is this something that can be achieved by manipulating the strides, or are there cases, when you actually have to copy data?

v923z, 2025-08-15:
@sfe-SparkFro I think this is addressed in https://github.com/v923z/micropython-ulab/pull/735. Do you think you could whip up the tests in all four dimensions, and all `dtype`s?