v923z/micropython-ulab #732: [FEATURE REQUEST]

kwagyeman, 2025-07-24:
Just would like for the % operator to be supported. 

I've had to do this instead since it is not supported:

```
def mod(a, b):
    return a - (b * (a // b))
```

It looks like it just needs to be added as one of the operators in the code base. Happy to send a PR for this, or, otherwise, just leaving it here as a feature request.

v923z, 2025-07-27:
Can you, please, check out https://github.com/v923z/micropython-ulab/pull/733, and see if you're satisfied with it? 

I'm a bit perplexed by `numpy`'s implementation, where I get the following:

```python
import numpy as np

for dtype_a in (np.uint8, np.int8, np.uint16, np.int16):
    a = np.array([1, 2, 3, 4], dtype=dtype_a)
    print(a.dtype)
    for dtype_b in (np.uint8, np.int8, np.uint16, np.int16):
        b = np.array([2, 3, 2, 4], dtype=dtype_b)
        print('====', b.dtype, (a % b).dtype)

```
```python

uint8
==== uint8 uint8
==== int8 int16
==== uint16 uint16
==== int16 int16
int8
==== uint8 int16
==== int8 int8
==== uint16 int32
==== int16 int16
uint16
==== uint8 uint16
==== int8 int32
==== uint16 uint16
==== int16 int32
int16
==== uint8 int16
==== int8 int16
==== uint16 int32
==== int16 int16
```
What bugs me a bit is that the results of the modulus should fit into the `dtype` of the left-hand-side operand, yet, `numpy` seems to promote the results, and I find that weird. Let me know what you want to do about it!

kwagyeman, 2025-07-27:
Uh, does this operation remalloc the array no matter what? If so, then follow numpy. Otherwise, if the array is modified in place then don't.

v923z, 2025-07-28:
> Uh, does this operation remalloc the array no matter what? If so, then follow numpy. Otherwise, if the array is modified in place then don't.

This is the implementation of `%`, so the result can be promoted. Do you also need `%=`? In that case the answer is clear: if the operation would need to promote the result, then an exception is raised in `numpy`.

kwagyeman, 2025-07-28:
Yeah, %= would be great!

v923z, 2025-08-04:
Implemented in https://github.com/v923z/micropython-ulab/pull/733, so closing now.

closed 2025-08-04 by v923z