v923z/micropython-ulab #737: [FEATURE REQUEST] Support for bincount?

kwagyeman, 2025-08-20:
https://numpy.org/doc/stable/reference/generated/numpy.bincount.html?

It solves a very challenging problem that otherwise has to be done in Python:

```
        # Reset the frequency histogram.
        freq_histogram[:, :, :] = 0

        # Flatten 3D index: (y, x, f) → idx
        flat_idx = (y_bins * X_BINS + x_bins) * FREQ_BINS + freq_indices

        # Update the frequency histogram.
        for i in range(len(target_events)):
            freq_histogram[flat_idx[i]] += gain[i]
```

kwagyeman, 2025-08-21:
It would be used like this:

```
# Keep ON or OFF events (use OR, not AND)
mask = (events_slice[:, 0] == csi.PIX_ON_EVENT) | (events_slice[:, 0] == csi.PIX_OFF_EVENT)
if np.any(mask):
    target_events = events_slice[mask]

    # +1 for ON, -1 for OFF (use ±64 if you want by multiplying at the end)
    gain = np.where(target_events[:, 0] == csi.PIX_ON_EVENT, 1, -1)

    # Timestamps → µs
    ts = target_events[:, 1] * 1000000 + target_events[:, 2] * 1000 + target_events[:, 3]

    # Frequency bin per event
    freq_indices = (ts * FREQ_BINS // 1000000) % FREQ_BINS

    # Spatial bins
    x_bins = (target_events[:, 4] * X_BINS) // 320
    y_bins = (target_events[:, 5] * Y_BINS) // 320

    # Flatten 3D index: (y, x, f) → idx
    flat_idx = (y_bins * X_BINS + x_bins) * FREQ_BINS + freq_indices

    # Zero the histogram quickly
    freq_histogram[:] = 0

    # Accumulate in flat space
    flat_len = Y_BINS * X_BINS * FREQ_BINS
    updates = np.bincount(flat_idx, weights=gain, minlength=flat_len)

    # Write back into 3D histogram
    freq_histogram.reshape((Y_BINS * X_BINS * FREQ_BINS,))[:] = updates
    freq_histogram.reshape((Y_BINS, X_BINS, FREQ_BINS))
```

1. Okay
2.  Okay
3. Yes
4. Yes

v923z, 2025-09-03:
This has been implemented in https://github.com/v923z/micropython-ulab/pull/738, so closing now.